alexCajas/esp8266RTOSArduCore #1: Err when build example

stingdau1206, 2024-04-30:
## Environment

- Development Kit: esp8266-nodemcu
- IDF version: 1.0.3
- Development Env:      Arduino IDE
- Operating System:    Ubuntu
- Power Supply:         USB

## Problem Description
after follow your step to setup & install to arduino ide on ubuntu, when build prj, i meet some err
```
preCompile Build Path: /tmp/arduino/sketches/F9C5C29705F1B054F458AC71338804F2
open /tmp/3625109955/sketch_merged.cpp: no such file or directory

Compilation error: open /tmp/3625109955/sketch_merged.cpp: no such file or directory
```
[image]

alexCajas, 2024-05-01:
Hi @stingdau1206 Thanks for your report!. It seems there is some problem with Arduino IDE 2.x, Can you try with Arduino IDE 1.8.x and let me know if it works well?

Jookia, 2024-07-15:
I get this on arduino-cli  Version: 0.35.3-arch Commit: 95cfd654fe0dd6b07e903d3132c0faceabfbe9e3

woshilaiba, 2025-02-04:
> I get this on arduino-cli Version: 0.35.3-arch Commit: 95cfd654fe0dd6b07e903d3132c0faceabfbe9e3

running arduino-cli 0.24  ok