alexCajas/esp8266RTOSArduCore #3: "esp8266RTOS\1.0.3/tools/createPreproc.py", line 7 print(f"preCompile Build Path: {build_path}", file=sys.stderr)

gunt2551, 2024-05-18:
## Environment

- Development Kit: ESP8266 wemos d1 mini pro
- IDF version : 1.0.3
- Development Env: Arduino IDE 1.8.19
- Operating System: Windows 11
- Power Supply: USB

## Problem Description
When i verifying code(your example code on github) I got an error. I am unsure the "build_path" needs to be  set at "Environment Variables" or is already set up on your code. where can i find the path?

I also tried with version 1.0.2 ,but I got the same error.

If this issue needs to be discussed on your community forum, please let me know.

>   File "C:\Users\user\AppData\Local\Arduino15\packages\esp8266RTOS\hardware\esp8266RTOS\1.0.3/tools/createPreproc.py", line 7
    print(f"preCompile Build Path: {build_path}", file=sys.stderr)
.....................................................................................^
SyntaxError: invalid syntax
exit status 1
Error compiling for board ESP8266 Module.

[image]

alexCajas, 2024-05-20:
Hi @gunt2551, after examining ther error message, I believe that the issue originates from your python version. Currently, I'm using 3.7.4. Normaly It woldn't be a problem, but sometimes there can be issues with backward compatibility. Could you try with this version and comment here if the problem persists?

gunt2551, 2024-05-21:
Thx @alexCajas. After changing the default Python version 2.7.18 called by "python" in the command line to 3.7.4 or 3.12.3, these versions can now run in the MSYS2 terminal. (Now, i use python 3.7.4 that the @alexCajas suggested me)

And when it runs in terminal, I need to fix the path from `backslash (\)` to `forward slash (/)` in the file that displayed the path in terminal before flashing (this process I need to do everytime when run the code)

[image]

and then it can flash successfully

This is for anyone who encountered the error.
 
> raise SerialException("could not open port {!r}: {!r}".format(self.portstr,ctypes.WinError()))
> serial.serialutil.SerialException: could not open port 'COM7': WindowsError(5, 'Access is denied.')
> esptool.py failed with exit code 1

**Close the `Serial monitor`** and then run it again

[image]